gonium/gosdm630 #7: Duplication of ID in Web API

gonium, 2017-04-27:
After talking to @TheCount, I merged the pull request https://github.com/gonium/gosdm630/pull/5 conditionally. The pull requests introduces a ``UniqueID`` which can be used to identify Modbus Device and the gosdm630 instance. A commandline argument allows a user to specify a prefix. The Modbus ID of the attached device will then be appended to this prefix.

I understand the use case and like the idea of a user-specified "tag" for the gosdm630 instance. But the device ID is also present in the ``ModbusID`` field - a client could simply concatenate the two to generate the same information. 

For the sake of keeping the API clean: Refactor this later so that the device ID is not appended to the ``UniqueID``. The name of the field should also be changed to ``InstanceID``. This needs to be revisited when @TheCount has more time.

andig, 2018-05-12:
ping @TheCount I'm recently helping @gonium with maintaining this repo. I've also used the `UniqueId` for topic generation in the mqtt client for consistency. Do you see need for further changes?

TheCount, 2018-06-08:
Personally, no, I can live with the status quo.

The issue here is that it would be sufficient (thus more minimalistic, thus cleaner) to have the ID merely identify the bus, not the instrument.

The thing is, software using the output of gosdm would need an extra operation to generate a unique instrument ID then. A perfectly reasonable requirement for software you are currently developing yourself; more onerous for legacy or third-party software with limited JSON-interpretation capabilities.

andig, 2018-06-08:
> The issue here is that it would be sufficient (thus more minimalistic, thus cleaner) to have the ID merely identify the bus, not the instrument.

Interesting point, i.e. in case of multiple modbus adapters?

Its still debatable then if the physical modbus layout (adapter, device numbers) should equal the logical API view.

closed 2019-01-13 by andig