gonium/gosdm630 #10: Control abort conditions

TheCount, 2017-06-30:
Currently, gosdm aborts after 5 consecutive failures. This is problematic in cases where several meters are connected to the bus, one of the meters fails, but one wants to continue logging data from the remaining meters.

It would therefore be nice if there was a flag/mode that, in case of one failure, would "forward" the scheduler to the next device ID in the hope that this device is still working.

gonium, 2017-07-12:
The software has only one error handling method right now: if a meter is not reachable, terminate the software. This simplifies the design significantly, but does not handle single-device failures well. The HTTP interface needs significant changes in order to do this:

1. For the ``/last/{id}`` calls, the HTTP handler should not send invalid data, but signal the current device state using the HTTP error code "503 Service Unavailable".
2. This will not work for the ``/last`` call. Instead, it should reflect the device state using some kind of "health" information. It is the callers responsibility to check the health state before processing the information. All readings of a failed smart meter should be set to ``null``.
3. As for the firehose: it should send the health information as a part of the datastream. If a meter is healthy, all other readings should be sent. If not, only the health information will be sent, followed by the next meter's readings.
4. The builtin website needs to evaluate the health information from the firehose.

Internally, there is no abstraction for a meter yet, i.e. there is no place to store the health information. In addition, different types of smart meters cannot be supported by the current code (see also issue #11). A simple list of meter metadata must be added to the code. If a modbus failure is detected, the meter should be marked accordingly.
The scheduler must incorporate the meter state into the request schedule - i.e. run a probe for all inactive meters, but if it fails, don't ask for more measurements.

All in all: This is a reasonable request, but the impact on the code is rather high.

gonium, 2017-07-24:
Commit 1eb4f0346 implements the status information over the http api, except for the firehose. The semantic of the initial design has changed a little. More tests and documentation needed.

gonium, 2017-07-27:
Now fully implemented and documentation updated. Closing.

closed 2017-07-27 by gonium