gonium/gosdm630 #18: since latest commit realtime view not working anymore

holli73, 2017-10-26:
i was on a version below 0.5 and all was fine - than i did a git pull and updated to the latest commit's today now my system ist not starting because of the changed -d parameter - i figured this out but i still do not get data in the realtime view - so i tried the sdm detect:

```
pi@rasp-eg-verteilerkasten ~ $ gosdm630/bin/sdm_detect
2017/10/26 09:05:00 Starting bus scan
2017/10/26 09:05:00 Device 1: SDM type device found, L1 voltage: [%!f(uint8=67) %!f(uint8=102) %!f(uint8=28) %!f(uint8=153)]
```
this looks already a bit strange?

if i start the http process i receive:

```
/home/pi/gosdm630/bin/sdm630_httpd -s /dev/rs485 -d sdm:1 -c 2 -f HolliHausGesamt[id-%d]
2017/10/26 08:58:14 Starting API httpd at :8080
2017/10/26 08:59:21 Device 1 failed to respond - retry attempt 1 of 5
2017/10/26 09:03:41 Device 1 failed to respond - retry attempt 1 of 5
2017/10/26 09:03:45 Device 1 failed to respond - retry attempt 1 of 5
```

if i start in verbose mode it looks like data is received - even if i do my rest call i get data - only the webview is not showing anything?

```
pi@rasp-eg-verteilerkasten ~ $ /home/pi/gosdm630/bin/sdm630_httpd -s /dev/rs485 -d sdm:1 -c 2 -f HolliHausGesamt[id-%d] -v
2017/10/26 09:12:44 Connecting to RTU via /dev/rs485, 9600 8N1
2017/10/26 09:12:44 Starting API httpd at :8080
2017/10/26 09:12:44 SubscriptionManager: Starting run.
RTUClientHandler: 2017/10/26 09:12:44 modbus: sending "\x01\x04\x00\x00\x00\x02q\xcb"
RTUClientHandler: 2017/10/26 09:12:44 modbus: received 01 04 04 43 67 66 36 f4 69
RTUClientHandler: 2017/10/26 09:12:44 modbus: sending "\x01\x04\x00\x02\x00\x02\xd0\v"
2017/10/26 09:12:44 Creating new eventBuffer for category: meterupdate
2017/10/26 09:12:44 SubscriptionManager: queued event: &{1509001964398 meterupdate DevID: 1, FunCode: 4, Opcode 0: Value: 231.399}.
2017/10/26 09:12:44 UniqueId: HolliHausGesamt[id-1] ID: 1 T: 2017-10-26T09:12:44+02:00 - L1: 231.40V NaNA NaNW NaNcos | L2: NaNV NaNA NaNW NaNcos | L3: NaNV NaNA NaNW NaNcos
RTUClientHandler: 2017/10/26 09:12:44 modbus: received 01 04 04 43 62 80 96 ae 70
RTUClientHandler: 2017/10/26 09:12:44 modbus: sending "\x01\x04\x00\x04\x00\x020\n"
2017/10/26 09:12:44 UniqueId: HolliHausGesamt[id-1] ID: 1 T: 2017-10-26T09:12:44+02:00 - L1: 231.40V NaNA NaNW NaNcos | L2: 226.50V NaNA NaNW NaNcos | L3: NaNV NaNA NaNW NaNcos
2017/10/26 09:12:44 SubscriptionManager: queued event: &{1509001964429 meterupdate DevID: 1, FunCode: 4, Opcode 2: Value: 226.502}.
RTUClientHandler: 2017/10/26 09:12:44 modbus: received 01 04 04 43 65 9f 8b d7 88
RTUClientHandler: 2017/10/26 09:12:44 modbus: sending "\x01\x04\x00\x06\x00\x02\x91\xca"
2017/10/26 09:12:44 UniqueId: HolliHausGesamt[id-1] ID: 1 T: 2017-10-26T09:12:44+02:00 - L1: 231.40V NaNA NaNW NaNcos | L2: 226.50V NaNA NaNW NaNcos | L3: 229.62V NaNA NaNW NaNcos
2017/10/26 09:12:44 SubscriptionManager: queued event: &{1509001964463 meterupdate DevID: 1, FunCode: 4, Opcode 4: Value: 229.623}.
RTUClientHandler: 2017/10/26 09:12:44 modbus: received 01 04 04 41 09 83 03 1e 8b
RTUClientHandler: 2017/10/26 09:12:44 modbus: sending "\x01\x04\x00\b\x00\x02\xf0\t"
2017/10/26 09:12:44 SubscriptionManager: queued event: &{1509001964496 meterupdate DevID: 1, FunCode: 4, Opcode 6: Value: 8.594}.
2017/10/26 09:12:44 UniqueId: HolliHausGesamt[id-1] ID: 1 T: 2017-10-26T09:12:44+02:00 - L1: 231.40V 8.59A NaNW NaNcos | L2: 226.50V NaNA NaNW NaNcos | L3: 229.62V NaNA NaNW NaNcos
RTUClientHandler: 2017/10/26 09:12:44 modbus: received 01 04 04 40 ea 25 7f 94 c0
RTUClientHandler: 2017/10/26 09:12:44 modbus: sending "\x01\x04\x00\n\x00\x02Q\xc9"
2017/10/26 09:12:44 SubscriptionManager: queued event: &{1509001964530 meterupdate DevID: 1, FunCode: 4, Opcode 8: Value: 7.317}.
2017/10/26 09:12:44 UniqueId: HolliHausGesamt[id-1] ID: 1 T: 2017-10-26T09:12:44+02:00 - L1: 231.40V 8.59A NaNW NaNcos | L2: 226.50V 7.32A NaNW NaNcos | L3: 229.62V NaNA NaNW NaNcos
RTUClientHandler: 2017/10/26 09:12:44 modbus: received 01 04 04 40 4d 12 4f 33 07
RTUClientHandler: 2017/10/26 09:12:44 modbus: sending "\x01\x04\x00\f\x00\x02\xb1\xc8"
2017/10/26 09:12:44 SubscriptionManager: queued event: &{1509001964564 meterupdate DevID: 1, FunCode: 4, Opcode a: Value: 3.204}.
2017/10/26 09:12:44 UniqueId: HolliHausGesamt[id-1] ID: 1 T: 2017-10-26T09:12:44+02:00 - L1: 231.40V 8.59A NaNW NaNcos | L2: 226.50V 7.32A NaNW NaNcos | L3: 229.62V 3.20A NaNW NaNcos
RTUClientHandler: 2017/10/26 09:12:44 modbus: received 01 04 04 44 ec cb 81 b9 d1
RTUClientHandler: 2017/10/26 09:12:44 modbus: sending "\x01\x04\x00\x0e\x00\x02\x10\b"
2017/10/26 09:12:44 SubscriptionManager: queued event: &{1509001964595 meterupdate DevID: 1, FunCode: 4, Opcode c: Value: 1894.359}.
2017/10/26 09:12:44 UniqueId: HolliHausGesamt[id-1] ID: 1 T: 2017-10-26T09:12:44+02:00 - L1: 231.40V 8.59A 1894.36W NaNcos | L2: 226.50V 7.32A NaNW NaNcos | L3: 229.62V 3.20A NaNW NaNcos
RTUClientHandler: 2017/10/26 09:12:44 modbus: received 01 04 04 44 bd 58 6b 05 7f
2017/10/26 09:12:44 SubscriptionManager: queued event: &{1509001964628 meterupdate DevID: 1, FunCode: 4, Opcode e: Value: 1514.763}.
2017/10/26 09:12:44 UniqueId: HolliHausGesamt[id-1] ID: 1 T: 2017-10-26T09:12:44+02:00 - L1: 231.40V 8.59A 1894.36W NaNcos | L2: 226.50V 7.32A 1514.76W NaNcos | L3: 229.62V 3.20A NaNW NaNcos
RTUClientHandler: 2017/10/26 09:12:44 modbus: sending "\x01\x04\x00\x10\x00\x02p\x0e"
RTUClientHandler: 2017/10/26 09:12:44 modbus: received 01 04 04 43 f3 2a a6 81 29
RTUClientHandler: 2017/10/26 09:12:44 modbus: sending "\x01\x04\x00\x1e\x00\x02\x11\xcd"
2017/10/26 09:12:44 SubscriptionManager: queued event: &{1509001964660 meterupdate DevID: 1, FunCode: 4, Opcode 10: Value: 486.333}.
2017/10/26 09:12:44 UniqueId: HolliHausGesamt[id-1] ID: 1 T: 2017-10-26T09:12:44+02:00 - L1: 231.40V 8.59A 1894.36W NaNcos | L2: 226.50V 7.32A 1514.76W NaNcos | L3: 229.62V 3.20A 486.33W NaNcos
RTUClientHandler: 2017/10/26 09:12:44 modbus: received 01 04 04 3f 73 d9 6b 1c 34
RTUClientHandler: 2017/10/26 09:12:44 modbus: sending "\x01\x04\x00 \x00\x02p\x01"
2017/10/26 09:12:44 SubscriptionManager: queued event: &{1509001964697 meterupdate DevID: 1, FunCode: 4, Opcode 1e: Value: 0.953}.
2017/10/26 09:12:44 UniqueId: HolliHausGesamt[id-1] ID: 1 T: 2017-10-26T09:12:44+02:00 - L1: 231.40V 8.59A 1894.36W 0.95cos | L2: 226.50V 7.32A 1514.76W NaNcos | L3: 229.62V 3.20A 486.33W NaNcos
RTUClientHandler: 2017/10/26 09:12:44 modbus: received 01 04 04 3f 69 fa 12 e5 21
RTUClientHandler: 2017/10/26 09:12:44 modbus: sending "\x01\x04\x00\"\x00\x02\xd1\xc1"
[... 322 lines cut ...]
RTUClientHandler: 2017/10/26 09:12:47 modbus: sending "\x01\x04\x01d\x00\x021\xe8"
2017/10/26 09:12:47 SubscriptionManager: queued event: &{1509001967538 meterupdate DevID: 1, FunCode: 4, Opcode 162: Value: 0.000}.
2017/10/26 09:12:47 UniqueId: HolliHausGesamt[id-1] ID: 1 T: 2017-10-26T09:12:47+02:00 - L1: 231.40V 8.63A 1902.19W 0.95cos | L2: 226.50V 7.21A 1488.54W 0.91cos | L3: 229.62V 3.21A 487.79W 0.66cos
RTUClientHandler: 2017/10/26 09:12:47 modbus: received 01 04 04 00 00 00 00 fb 84
RTUClientHandler: 2017/10/26 09:12:47 modbus: sending "\x01\x04\x00J\x00\x02P\x1d"
2017/10/26 09:12:47 SubscriptionManager: queued event: &{1509001967574 meterupdate DevID: 1, FunCode: 4, Opcode 164: Value: 0.000}.
2017/10/26 09:12:47 UniqueId: HolliHausGesamt[id-1] ID: 1 T: 2017-10-26T09:12:47+02:00 - L1: 231.40V 8.63A 1902.19W 0.95cos | L2: 226.50V 7.21A 1488.54W 0.91cos | L3: 229.62V 3.21A 487.79W 0.66cos
RTUClientHandler: 2017/10/26 09:12:47 modbus: received 01 04 04 00 00 00 00 fb 84
RTUClientHandler: 2017/10/26 09:12:47 modbus: sending "\x01\x04\x00\xea\x00\x02P?"
2017/10/26 09:12:47 SubscriptionManager: queued event: &{1509001967608 meterupdate DevID: 1, FunCode: 4, Opcode 4a: Value: 0.000}.
2017/10/26 09:12:47 UniqueId: HolliHausGesamt[id-1] ID: 1 T: 2017-10-26T09:12:47+02:00 - L1: 231.40V 8.63A 1902.19W 0.95cos | L2: 226.50V 7.21A 1488.54W 0.91cos | L3: 229.62V 3.21A 487.79W 0.66cos
RTUClientHandler: 2017/10/26 09:12:47 modbus: received 01 04 04 40 4a 1e ac c6 4f
RTUClientHandler: 2017/10/26 09:12:47 modbus: sending "\x01\x04\x00\xec\x00\x02\xb0>"
2017/10/26 09:12:47 SubscriptionManager: queued event: &{1509001967646 meterupdate DevID: 1, FunCode: 4, Opcode ea: Value: 3.158}.
2017/10/26 09:12:47 UniqueId: HolliHausGesamt[id-1] ID: 1 T: 2017-10-26T09:12:47+02:00 - L1: 231.40V 8.63A 1902.19W 0.95cos | L2: 226.50V 7.21A 1488.54W 0.91cos | L3: 229.62V 3.21A 487.79W 0.66cos
RTUClientHandler: 2017/10/26 09:12:47 modbus: received 01 04 04 3f 7e 89 bb b0 6b
RTUClientHandler: 2017/10/26 09:12:47 modbus: sending "\x01\x04\x00\xee\x00\x02\x11\xfe"
2017/10/26 09:12:47 SubscriptionManager: queued event: &{1509001967683 meterupdate DevID: 1, FunCode: 4, Opcode ec: Value: 0.994}.
2017/10/26 09:12:47 UniqueId: HolliHausGesamt[id-1] ID: 1 T: 2017-10-26T09:12:47+02:00 - L1: 231.40V 8.63A 1902.19W 0.95cos | L2: 226.50V 7.21A 1488.54W 0.91cos | L3: 229.62V 3.21A 487.79W 0.66cos
^C
```
any idea?

thanks
holli

andig, 2017-10-27:
The formatting errors are also reported here: https://github.com/gonium/gosdm630/issues/15

andig, 2017-10-27:
@holli73 lustiger Raspiname- meiner heisst uv wie Unterverteilung ;)