gonium/gosdm630 #20: High load on raspberry

andig, 2017-11-20:
I have a raspberry pi3 with two SDM630 attached via USB connector on one bus running at 9600 baud. When the http daemon is running the load is `0.65` when a single browser window is connected load is `1.25`.

This seems rather high given that the pi3 is quite capable?

miroslavpetrov, 2017-12-09:
I am currently testing on Raspberry Pi 1 Model B+ and cpu usage is 75% only by sdm630_httpd.

dimagoltsman, 2017-12-27:
im with RPI3 and besides the high cpu, it uses 900 MB of memory and stops responding every few days. is there a way to just get single measurement without starting the server?

MatzeJoerling, 2018-04-04:
RPI A, RS485 shield, nodered querying every 2 sec, no browser, => load 0.25

ABCarnage, 2018-05-03:
Had to put baudrate to 2400 baud to get 4 seconds latency (tested by pulling out devices). 9600 baud gave me a whopping 20 seconds. (using RaPi Model B 1, so yeah. But it works.)

andig, 2018-05-08:
Imho the problem is not the http daemon but the fact that the longpoll library constantly hits the httpd. I'm working on moving from longpoll to websockets which should improve the load.

andig, 2018-05-08:
Closing as duplicate of #13

closed 2018-05-08 by andig

andig, 2018-05-10:
See https://github.com/gonium/gosdm630/pull/38 for a websocket implementation- i would appreciate test feedback