gonium/gosdm630 #28: raspi cannot read sdm when rs485 is 1st serial

MatzeJoerling, 2018-04-04:
searching some time for that, if you rs485 is the 1st serial in your raspi the kernel / init starts a console on it. 
after that no real comminication with sdm is possible.
i  had to remove the console serial stuff from /boot/cmdline.txt

Only for the docs

RubberDucky007, 2019-03-16:
Does this affect only the real serial console (on GPIO) or 'console=tty1' in cmdline.txt too? My RS-485 adapter is /dev/ttyUSB0.

andig, 2019-03-16:
> Does this affect only the real serial console (on GPIO) or 'console=tty1' in cmdline.txt too? My RS-485 adapter is /dev/ttyUSB0.

Do you have any problem with `Zdev/ttyUSB0`? Working for me.

RubberDucky007, 2019-03-16:
No Problems so far i know. But thinking about the posibility this causes Problems getting all Values from the meter. Working with 24008N1 or 24008E1 all the same. L3=L2 and some doubled (calculated) so it can't be useful.