gonium/gosdm630 #29: Device 9 failed to respond - retry attempt 1 of 5

MatzeJoerling, 2018-04-04:
my SDM is device 9, when starting the webserver i got this every second:

```
2018/04/04 19:51:15 Device 9 failed to respond - retry attempt 1 of 5
2018/04/04 19:51:15 Device 9 failed to respond - retry attempt 1 of 5
2018/04/04 19:51:16 Device 9 failed to respond - retry attempt 1 of 5
2018/04/04 19:51:16 Device 9 failed to respond - retry attempt 1 of 5
2018/04/04 19:51:17 Device 9 failed to respond - retry attempt 1 of 5
2018/04/04 19:51:18 Device 9 failed to respond - retry attempt 1 of 5
2018/04/04 19:51:18 Device 9 failed to respond - retry attempt 1 of 5
```
precompiled binary, release 0.5.1
./sdm630_httpd-linux-arm -s /dev/ttyAMA0  -d sdm:9

Any ideas?

regards M.

MatzeJoerling, 2018-04-04:
[error.log](https://github.com/gonium/gosdm630/files/1876953/error.log)

Error Log, webserver started with  "-v"

andig, 2018-05-12:
Looks similar to https://github.com/gonium/gosdm630/issues/18 - is this still a problem?

andig, 2018-09-25:
Closing as inactive.

closed 2018-09-25 by andig