gonium/gosdm630 #30: RS485 Board

preussal, 2018-04-28:
Hi,
do you think that this
[Raspberry Pi Shield - RS485-Schnittstelle](https://www.reichelt.de/Entwicklerboard-Zubehoer/RPI-RS485/3/index.html?ACTION=3&GROUPID=8244&ARTICLE=162304) for ~ 10€ 
or this
[RS422 / RS485 Shield für Raspberry Pi](https://www.hwhardsoft.de/deutsch/projekte/rs485-shield/) for ~ 30€
would work too.

Grüße aus München

ABCarnage, 2018-05-03:
I use a USB-RS485 bridge with 3 ports (A,B,GND) from China which cost so much less than your options. But it should work. 
Hint: Raspberry Pi can't get the requests out fast enough when on 9600 baud (20 seconds latency). We chose for 2400 baud which gives about 4 seconds latency.

andig, 2018-09-25:
I've been using InCircuit https://shop.in-circuit.de/product_info.php?products_id=81 for ages and its working stable with two Eastron meters.

closed 2018-09-25 by andig