gonium/gosdm630 #37: Add SDM120 meter type

andig, 2018-05-08:
As per https://github.com/mreschka/gosdm630/commit/6c82a0be70c2626fcb5ee395712cd66e2b898c4c and https://github.com/mreschka/gosdm630/commit/12dabddff6e3ab850496d0a933a4d4252dd6f8a8

andig, 2018-05-14:
ping @mreschka I'd like to understand if adding SDM120 to this repo would be a good idea. Could you kindly clarify what drawbacks there are using the existing SDM630 meter for SDM120 devices? Are there any:

  - logfile errors (due to failing meter queries)?
  - UI problems with the web ui (empty rows)?
  - API problems (improper handling of null values)?
  - other problems?

andig, 2018-09-21:
ping @mreschka

mreschka, 2018-10-07:
Hi @andig,

sorry, I forgot to respond to your first message, thanks for the ping. 

No there are no problems with the sdm630 version für a 120 device. The motivation is performance. The sdm120 supports far less registers and I reduced the scheduler to theses ones only. I have in total 20 sdm120 meters and this boosts the update-rate more than 2 times.

Viele Grüße

Markus

andig, 2018-10-07:
Hi Markus,

Thanks for the update, I can certainly see your use case. You're trying to push update rate as high as possible?

I'm looking into adding additional meters and e.g. inverters (https://github.com/gonium/gosdm630/pull/80) which would require to refactor the reading logic. I'm thinking to do block-wise register reading instead of single by single. In https://github.com/gonium/gosdm630/issues/46 I had in mind to do this automatically but #80 suggests that should be done in the meter definition. 

If I get around to actually implement something- would you be interested in testing?

andig, 2018-10-10:
@mreschka see https://github.com/gonium/gosdm630/pull/83 for an approach to do batch readings. Still needs to be implemented for other meter types.

J-A-U, 2018-12-09:
> If I get around to actually implement something- would you be interested in testing?

Sure.

andig, 2018-12-10:
@J-A-U its not a priority, I'll focus on finishing 0.8 for time being. The infrastructure for batch readings is already implemented though and will lend itself to improve readout speed for any meter type.