gonium/gosdm630 #45: mqtt Power(Watt) total and 1messageblock every 30sec?

timtimsson, 2018-05-31:
Hello,
i have 2 Question about your  great v.0.6.0
1. i use sdm630 with mqtt. imo mqtt stress my rasperry - is it possible to create one messageblock every 30sec or 1min?
2.  i miss WLocPhs total - please can you integrate this in mqtt?

Regards Tim

```

/Raspi/SDM630/Instrument1/WLocPhsA 115.479
/Raspi/SDM630/Instrument1/WLocPhsB 80.371
/Raspi/SDM630/Instrument1/WLocPhsC 167.294
/Raspi/SDM630/Instrument1/AngLocPhsA 0.717
/Raspi/SDM630/Instrument1/AngLocPhsB 0.538
/Raspi/SDM630/Instrument1/AngLocPhsC 0.988
/Raspi/SDM630/Instrument1/TotkWhImportPhsA 0.329
/Raspi/SDM630/Instrument1/TotkWhImportPhsB 0.548
/Raspi/SDM630/Instrument1/TotkWhImportPhsC 1.156
/Raspi/SDM630/Instrument1/TotkWhImport 2.033
```

andig, 2018-05-31:
> i miss WLocPhs total - please can you integrate this in mqtt?

Actually- at least for the SDM devices- that register is not defined in the source code and I wouldn't know if the device event supports it. The web UI solves this by just adding the phase powers.

I'd prefer to keep the MQTT API with what is really read from the device. So either we need the proper register or you need to implement this on client side.

> i use sdm630 with mqtt. imo mqtt stress my rasperry - is it possible to create one messageblock every 30sec or 1min?

We already have the rate parameter for that purpose:

		cli.IntFlag{
			Name:  "rate, r",
			Value: 0,
			Usage: "MQTT: The maximum update rate (default 0, i.e. unlimited) (after a push we will ignore more data from same device and channel for this time)",
			// Destination: &mqttRate,
		},

See https://github.com/gonium/gosdm630/issues/46 for further performance discussion.

andig, 2018-05-31:
Actually, just found that https://bg-etech.de/download/manual/SDM630Register.pdf documents quite a number of other registers that are not implemented. Wondering which ones really make sense?

timtimsson, 2018-05-31:
... oh, now i see that 
> <td>${ (parseFloat(meter.WLocPhsA) + parseFloat(meter.WLocPhsB) +parseFloat(meter.WLocPhsC)).toFixed(2) }</td>

-r 30 works perfect for me 
Thanks!

closed 2018-05-31 by timtimsson

andig, 2018-05-31:
ping @timtimsson would appreciate if you could test https://github.com/gonium/gosdm630/pull/47