gonium/gosdm630 #55: Add new precompiled release

dimagoltsman, 2018-07-25:
can u please add precompiled version? all day im sitting with a raspberry pi, it just wont compile....

andig, 2018-07-25:
Planning to do so; however you should be able to do this yourself:

  - on the raspy run `make binaries`
  - to cross-compile on another machine run `make release`

Could you share how exactly it is not compiling for you?

dimagoltsman, 2018-07-25:
```
pi@raspberrypi:~/go/src/gosdm630 $ make
Generating embedded assets
Embedded  /css/app.css
Embedded  /css/bootstrap.min.css
Embedded  /index.tmpl
Embedded  /js/app.js
Embedded  /js/bootstrap.min.js
Embedded  /js/jquery-3.2.1.min.js
Embedded  /js/tether.min.js
Embedded  /js/vue.min.js
Embedded  /simple.tmpl
Generation complete in 1.452229077s
Building for host platform
# github.com/gonium/gosdm630
../github.com/gonium/gosdm630/http.go:26:12: undefined: GetEmbeddedContent
../github.com/gonium/gosdm630/http.go:241:3: undefined: GetEmbeddedContent
Makefile:10: recipe for target 'binaries' failed
make: *** [binaries] Error 2
```

andig, 2018-07-25:
Ok, it seems you have `embed` installed and in the path already. 
Do you also have `dep` installed? If not please do: `go get -u github.com/golang/dep/cmd/dep` and do a `dep ensure`to install all vendor dependencies.

dimagoltsman, 2018-07-25:
same...

```
pi@raspberrypi:~/go/src/gosdm630 $ make
Generating embedded assets
Embedded  /css/app.css
Embedded  /css/bootstrap.min.css
Embedded  /index.tmpl
Embedded  /js/app.js
Embedded  /js/bootstrap.min.js
Embedded  /js/jquery-3.2.1.min.js
Embedded  /js/tether.min.js
Embedded  /js/vue.min.js
Embedded  /simple.tmpl
Generation complete in 1.427360206s
Building for host platform
# gosdm630/vendor/github.com/gonium/gosdm630
vendor/github.com/gonium/gosdm630/http.go:26:12: undefined: GetEmbeddedContent
vendor/github.com/gonium/gosdm630/http.go:241:3: undefined: GetEmbeddedContent
Makefile:10: recipe for target 'binaries' failed
make: *** [binaries] Error 2

```

andig, 2018-07-25:
Could you do an ls in the gosdm630 folder? Is there an embeddedcontent.go? If not- anywhere else?

dimagoltsman, 2018-07-25:
no, its just a clone of this repom=, there is no such file

andig, 2018-07-25:
Mhhm. Then ˋembedˋ fails to generate the fail although it says it did. remains cross-compiling as option on another machine. Go produces static binaries without external dependencies.

dimagoltsman, 2018-07-25:
so ill be glad if you will upload a compiled release.... i am getting the same error on osx

andig, 2018-07-26:
I've attached a compiled raspi binary for you, based on master branch plus https://github.com/gonium/gosdm630/pull/52. Could you check if this is working for you? If ok I'd like to merge and release.

[sdm630-linux-arm.zip](https://github.com/gonium/gosdm630/files/2230725/sdm630-linux-arm.zip)

dimagoltsman, 2018-07-26:
the server is running, but i cant get it to recognise the device

```
pi@raspberrypi:/home/homeassistant/.homeassistant $ /home/pi/gosdm630/bin/sdm_detect 
2018/07/26 09:55:15 Starting bus scan
2018/07/26 09:55:15 Device 1: JANITZA type device found, L1 Voltage (V): 0.00
2018/07/26 09:55:15 Device 2: n/a
2018/07/26 09:55:15 Device 3: n/a
2018/07/26 09:55:16 Device 4: n/a

```

```
pi@raspberrypi:/home/homeassistant/.homeassistant $  /home/pi/gosdm630/bin/sdm630_httpd  -s /dev/ttyUSB0 -u 0.0.0.0:63080
2018/07/26 09:56:31 Starting API httpd at 0.0.0.0:63080
2018/07/26 09:56:31 Device 1 failed to respond - retry attempt 1 of 5
2018/07/26 09:56:31 Device 1 failed to respond - retry attempt 2 of 5
2018/07/26 09:56:32 Device 1 failed to respond - retry attempt 3 of 5
2018/07/26 09:56:32 Device 1 failed to respond - retry attempt 4 of 5
2018/07/26 09:56:32 Device 1 failed to respond - retry attempt 5 of 5
2018/07/26 09:56:32 Device 1 failed to respond - retry attempt 1 of 5
2018/07/26 09:56:32 Failure - deactivating meter 1: Device 1 did not respond.
2018/07/26 09:56:32 Device 1 failed to respond - retry attempt 2 of 5
2018/07/26 09:56:32 Device 1 failed to respond - retry attempt 3 of 5
2018/07/26 09:56:32 Device 1 failed to respond - retry attempt 4 of 5
2018/07/26 09:56:32 Device 1 failed to respond - retry attempt 5 of 5
2018/07/26 09:56:32 Device 1 failed to respond - retry attempt 1 of 5
2018/07/26 09:56:32 Failure - deactivating meter 1: Device 1 did not respond.
2018/07/26 09:56:32 Device 1 failed to respond - retry attempt 2 of 5
2018/07/26 09:56:33 Device 1 failed to respond - retry attempt 3 of 5
2018/07/26 09:56:33 Device 1 failed to respond - retry attempt 4 of 5
2018/07/26 09:56:33 Device 1 failed to respond - retry attempt 5 of 5
2018/07/26 09:56:33 Failure - deactivating meter 1: Device 1 did not respond.
2018/07/26 09:56:34 http: response.Write on hijacked connection
2018/07/26 09:56:36 http: multiple response.WriteHeader calls
2018/07/26 09:56:36 http: multiple response.WriteHeader calls
2018/07/26 09:56:37 http: multiple response.WriteHeader calls
2018/07/26 09:56:41 http: multiple response.WriteHeader calls
2018/07/26 09:56:42 http: multiple response.WriteHeader calls
2018/07/26 09:56:43 http: response.Write on hijacked connection
2018/07/26 09:56:47 http: multiple response.WriteHeader calls
2018/07/26 09:56:47 http: multiple response.WriteHeader calls
```

andig, 2018-07-26:
What kind of device do you actually have attached? detect says a Janitza (though the voltage looks wrong if the device is actually attached to power) yet you're running httpd with default parameters which imply an SDM device.

Did you ever get the currently released version properly running before?

andig, 2018-07-26:
Ok, thats an SDM220, so http should work. Apparently I broke something during that last commits. I‘ll upload an older binary for you.

dimagoltsman, 2018-07-26:
You are awesome!

andig, 2018-07-26:
Here you are- this is the current master without #52 

[sdm630-linux-arm.zip](https://github.com/gonium/gosdm630/files/2231779/sdm630-linux-arm.zip)

dimagoltsman, 2018-07-26:
Perfect! works with same exec commands as the old one (i have a daemon for it). already integrated to HomeAssistant and added frequency to my sensors.
https://imgur.com/sk50Rw8
thank you!

andig, 2018-07-26:
GreatI've published the release: https://github.com/gonium/gosdm630/releases/tag/v0.6.0

Could you kindly double-check what `sdm_detect` outputs with the version you currently have installed?

dimagoltsman, 2018-07-26:
```
pi@raspberrypi:~ $ /home/pi/gosdm630/bin/sdm_detect 
2018/07/26 15:22:35 Starting bus scan
2018/07/26 15:22:35 Device 1: DZG type device found, L1 voltage: 0.00
2018/07/26 15:22:35 Device 2: n/a
2018/07/26 15:22:35 Device 3: n/a
2018/07/26 15:22:36 Device 4: n/a
2018/07/26 15:22:36 Device 5: n/a
2018/07/26 15:22:36 Device 6: n/a
2018/07/26 15:22:37 Device 7: n/a
2018/07/26 15:22:37 Device 8: n/a
2018/07/26 15:22:37 Device 9: n/a
```

still wrong voltage

andig, 2018-07-26:
> oh its probably conflict. when i shut down http server, it seems ok

oh, I see. Would you mind double-checking if detect is working in the binary that I first posted?

dimagoltsman, 2018-07-26:
no, doesnt work. it recognizes JANITZA and voltage 0.0. (turned off http first).
the new one works fine