gonium/gosdm630 #59: Release 0.6.0 mqtt disconnected

timtimsson, 2018-09-02:
Hi,
i use Release 0.6.0 with mqtt and it works - thanks... !
If i checked mqtt history and i can only find disconnected - i miss "connected" 
Thanks!

```
Pi:~ $ mosquitto_sub -v -t "/Raspi/SDM630/status/#" 
/Raspi/SDM630/status disconnected
/Raspi/SDM630/status disconnected
/Raspi/SDM630/status disconnected
/Raspi/SDM630/status disconnected
/Raspi/SDM630/status disconnected
/Raspi/SDM630/status disconnected
```

andig, 2018-09-18:
Ok, so `disconnected` works and is published by the broker itself as "will": https://github.com/gonium/gosdm630/blob/master/mqtt.go#L75

As for `connected` it seems that this line https://github.com/gonium/gosdm630/blob/master/mqtt.go#L99 needs to move one down- we're apparently publishing the `disconnected` message twice instead of `connected`. 

Could you check if swapping these lines fixes the problem?

timtimsson, 2018-09-18:
... i change the 2rows and it works - fine
But i have a error with the webgui - if i clear my browser-cache - no data reseive...., but mqtt 

[image]

andig, 2018-09-18:
Any error in the browser console?

closed 2018-09-18 by andig

andig, 2018-09-19:
> maybe - how can i fix that

Fixing always starts with diagnosis. So- *what* error are you seeing in the console upon reloading the page?

timtimsson, 2018-09-19:
` -v
2018/09/19 08:53:52 Connecting to RTU via /dev/ttyUSB1, 9600 8N1
2018/09/19 08:53:52 Connecting MQTT at tcp://10.1.1.200:1883
2018/09/19 08:53:52     clientid:     sdm630
2018/09/19 08:53:52     user:         Broker_ESP
2018/09/19 08:53:52     password:     ****
2018/09/19 08:53:52     topic:        /Raspi/SDM630
2018/09/19 08:53:52     qos:          1
2018/09/19 08:53:52     cleansession: false
2018/09/19 08:53:52 MQTT: connected
2018/09/19 08:53:52 MQTT: push /Raspi/SDM630/status, message: connected
2018/09/19 08:53:52 Starting API httpd at :8081
RTUClientHandler: 2018/09/19 08:53:52 modbus: sending "\x01\x04\x00\x00\x00\x02q\xcb"
RTUClientHandler: 2018/09/19 08:53:52 modbus: received 01 04 04 43 69 3b 0d ec e9
RTUClientHandler: 2018/09/19 08:53:52 modbus: sending "\x01\x04\x00\x02\x00\x02\xd0\v"
2018/09/19 08:53:52 UniqueId: Instrument1 ID: 1 T: 2018-09-19T08:53:52+02:00 - L1: 233.23V NaNA NaNW NaNcos | L2: NaNV NaNA NaNW NaNcos | L3: NaNV NaNA NaNW NaNcos%!(EXTRA float64=NaN)
2018/09/19 08:53:52 MQTT: got meter data (device 1: data: VolLocPhsA, value: 233.231 W, desc: L1 Voltage (V))
2018/09/19 08:53:52 MQTT: push /Raspi/SDM630/Instrument1/VolLocPhsA, message: 233.231
RTUClientHandler: 2018/09/19 08:53:52 modbus: received 01 04 04 43 6a f0 9d 4b b5
RTUClientHandler: 2018/09/19 08:53:52 modbus: sending "\x01\x04\x00\x04\x00\x020\n"
2018/09/19 08:53:52 MQTT: got meter data (device 1: data: VolLocPhsB, value: 234.940 W, desc: L2 Voltage (V))
2018/09/19 08:53:52 MQTT: push /Raspi/SDM630/Instrument1/VolLocPhsB, message: 234.940
2018/09/19 08:53:52 UniqueId: Instrument1 ID: 1 T: 2018-09-19T08:53:52+02:00 - L1: 233.23V NaNA NaNW NaNcos | L2: 234.94V NaNA NaNW NaNcos | L3: NaNV NaNA NaNW NaNcos%!(EXTRA float64=NaN)
RTUClientHandler: 2018/09/19 08:53:52 modbus: received 01 04 04 43 6b aa 76 60 9a
RTUClientHandler: 2018/09/19 08:53:52 modbus: sending "\x01\x04\x00\x06\x00\x02\x91\xca"
2018/09/19 08:53:52 UniqueId: Instrument1 ID: 1 T: 2018-09-19T08:53:52+02:00 - L1: 233.23V NaNA NaNW NaNcos | L2: 234.94V NaNA NaNW NaNcos | L3: 235.67V NaNA NaNW NaNcos%!(EXTRA float64=NaN)
2018/09/19 08:53:52 MQTT: got meter data (device 1: data: VolLocPhsC, value: 235.666 W, desc: L3 Voltage (V))
2018/09/19 08:53:52 MQTT: push /Raspi/SDM630/Instrument1/VolLocPhsC, message: 235.666
RTUClientHandler: 2018/09/19 08:53:52 modbus: received 01 04 04 3e ec aa fe c8 b9
RTUClientHandler: 2018/09/19 08:53:52 modbus: sending "\x01\x04\x00\b\x00\x02\xf0\t"
2018/09/19 08:53:52 UniqueId: Instrument1 ID: 1 T: 2018-09-19T08:53:52+02:00 - L1: 233.23V 0.46A NaNW NaNcos | L2: 234.94V NaNA NaNW NaNcos | L3: 235.67V NaNA NaNW NaNcos%!(EXTRA float64=NaN)
2018/09/19 08:53:52 MQTT: got meter data (device 1: data: AmpLocPhsA, value: 0.462 W, desc: L1 Current (A))
2018/09/19 08:53:52 MQTT: push /Raspi/SDM630/Instrument1/AmpLocPhsA, message: 0.462
RTUClientHandler: 2018/09/19 08:53:52 modbus: received 01 04 04 3f 5a e1 1f de 1b
RTUClientHandler: 2018/09/19 08:53:52 modbus: sending "\x01\x04\x00\n\x00\x02Q\xc9"
2018/09/19 08:53:52 MQTT: got meter data (device 1: data: AmpLocPhsB, value: 0.855 W, desc: L2 Current (A))
2018/09/19 08:53:52 MQTT: push /Raspi/SDM630/Instrument1/AmpLocPhsB, message: 0.855
2018/09/19 08:53:52 UniqueId: Instrument1 ID: 1 T: 2018-09-19T08:53:52+02:00 - L1: 233.23V 0.46A NaNW NaNcos | L2: 234.94V 0.85A NaNW NaNcos | L3: 235.67V NaNA NaNW NaNcos%!(EXTRA float64=NaN)
RTUClientHandler: 2018/09/19 08:53:53 modbus: received 01 04 04 3f cd 3c 07 36 ad
RTUClientHandler: 2018/09/19 08:53:53 modbus: sending "\x01\x04\x00\f\x00\x02\xb1\xc8"
2018/09/19 08:53:53 MQTT: got meter data (device 1: data: AmpLocPhsC, value: 1.603 W, desc: L3 Current (A))
2018/09/19 08:53:53 MQTT: push /Raspi/SDM630/Instrument1/AmpLocPhsC, message: 1.603
2018/09/19 08:53:53 UniqueId: Instrument1 ID: 1 T: 2018-09-19T08:53:53+02:00 - L1: 233.23V 0.46A NaNW NaNcos | L2: 234.94V 0.85A NaNW NaNcos | L3: 235.67V 1.60A NaNW NaNcos%!(EXTRA float64=NaN)
RTUClientHandler: 2018/09/19 08:53:53 modbus: received 01 04 04 42 51 f0 ae 7b 91
RTUClientHandler: 2018/09/19 08:53:53 modbus: sending "\x01\x04\x00\x0e\x00\x02\x10\b"
2018/09/19 08:53:53 MQTT: got meter data (device 1: data: WLocPhsA, value: 52.485 W, desc: L1 Power (W))
2018/09/19 08:53:53 MQTT: push /Raspi/SDM630/Instrument1/WLocPhsA, message: 52.485
2018/09/19 08:53:53 UniqueId: Instrument1 ID: 1 T: 2018-09-19T08:53:53+02:00 - L1: 233.23V 0.46A 52.49W NaNcos | L2: 234.94V 0.85A NaNW NaNcos | L3: 235.67V 1.60A NaNW NaNcos%!(EXTRA float64=NaN)
RTUClientHandler: 2018/09/19 08:53:53 modbus: received 01 04 04 43 1a c0 3a'

timtimsson, 2018-09-19:
sdm630_httpd:

`./sdm630_httpd -u :8081 -s /dev/ttyUSB1
2018/09/19 09:08:22 Starting API httpd at :8081
^C
pi@HomeServerPi:/usr/local/bin $ ./sdm630_httpd -u :8081 -s /dev/ttyUSB1 -v
2018/09/19 09:08:39 Connecting to RTU via /dev/ttyUSB1, 9600 8N1
2018/09/19 09:08:39 Starting API httpd at :8081
2018/09/19 09:08:39 SubscriptionManager: Starting run.
RTUClientHandler: 2018/09/19 09:08:39 modbus: sending "\x01\x04\x00\x00\x00\x02q\xcb"
RTUClientHandler: 2018/09/19 09:08:39 modbus: received 01 04 04 43 68 51 e7 12 06
2018/09/19 09:08:39 UniqueId: Instrument1 ID: 1 T: 2018-09-19T09:08:39+02:00 - L1: 232.32V NaNA NaNW NaNcos | L2: NaNV NaNA NaNW NaNcos | L3: NaNV NaNA NaNW NaNcos%!(EXTRA float64=NaN)
RTUClientHandler: 2018/09/19 09:08:39 modbus: sending "\x01\x04\x00\x02\x00\x02\xd0\v"
2018/09/19 09:08:39 Creating new eventBuffer for category: meterupdate
2018/09/19 09:08:39 SubscriptionManager: queued event: &{1537340919250 meterupdate DevID: 1, FunCode: 4, Opcode 0: Value: 232.320}.
RTUClientHandler: 2018/09/19 09:08:39 modbus: received 01 04 04 43 68 73 cf 0a b8
RTUClientHandler: 2018/09/19 09:08:39 modbus: sending "\x01\x04\x00\x04\x00\x020\n"
2018/09/19 09:08:39 UniqueId: Instrument1 ID: 1 T: 2018-09-19T09:08:39+02:00 - L1: 232.32V NaNA NaNW NaNcos | L2: 232.45V NaNA NaNW NaNcos | L3: NaNV NaNA NaNW NaNcos%!(EXTRA float64=NaN)
2018/09/19 09:08:39 SubscriptionManager: queued event: &{1537340919282 meterupdate DevID: 1, FunCode: 4, Opcode 2: Value: 232.452}.
RTUClientHandler: 2018/09/19 09:08:39 modbus: received 01 04 04 43 68 b0 47 5a 2e
RTUClientHandler: 2018/09/19 09:08:39 modbus: sending "\x01\x04\x00\x06\x00\x02\x91\xca"
2018/09/19 09:08:39 UniqueId: Instrument1 ID: 1 T: 2018-09-19T09:08:39+02:00 - L1: 232.32V NaNA NaNW NaNcos | L2: 232.45V NaNA NaNW NaNcos | L3: 232.69V NaNA NaNW NaNcos%!(EXTRA float64=NaN)
2018/09/19 09:08:39 SubscriptionManager: queued event: &{1537340919314 meterupdate DevID: 1, FunCode: 4, Opcode 4: Value: 232.689}.
RTUClientHandler: 2018/09/19 09:08:39 modbus: received 01 04 04 3e ea 01 6b 97 e7
RTUClientHandler: 2018/09/19 09:08:39 modbus: sending "\x01\x04\x00\b\x00\x02\xf0\t"
2018/09/19 09:08:39 UniqueId: Instrument1 ID: 1 T: 2018-09-19T09:08:39+02:00 - L1: 232.32V 0.46A NaNW NaNcos | L2: 232.45V NaNA NaNW NaNcos | L3: 232.69V NaNA NaNW NaNcos%!(EXTRA float64=NaN)
2018/09/19 09:08:39 SubscriptionManager: queued event: &{1537340919346 meterupdate DevID: 1, FunCode: 4, Opcode 6: Value: 0.457}.
RTUClientHandler: 2018/09/19 09:08:39 modbus: received 01 04 04 3f 16 d3 04 4a a7
2018/09/19 09:08:39 UniqueId: Instrument1 ID: 1 T: 2018-09-19T09:08:39+02:00 - L1: 232.32V 0.46A NaNW NaNcos | L2: 232.45V 0.59A NaNW NaNcos | L3: 232.69V NaNA NaNW NaNcos%!(EXTRA float64=NaN)
RTUClientHandler: 2018/09/19 09:08:39 modbus: sending "\x01\x04\x00\n\x00\x02Q\xc9"
2018/09/19 09:08:39 SubscriptionManager: queued event: &{1537340919378 meterupdate DevID: 1, FunCode: 4, Opcode 8: Value: 0.589}.
RTUClientHandler: 2018/09/19 09:08:39 modbus: received 01 04 04 3f cf 58 77 bc 49
RTUClientHandler: 2018/09/19 09:08:39 modbus: sending "\x01\x04\x00\f\x00\x02\xb1\xc8"
2018/09/19 09:08:39 SubscriptionManager: queued event: &{1537340919411 meterupdate DevID: 1, FunCode: 4, Opcode a: Value: 1.620}.
2018/09/19 09:08:39 UniqueId: Instrument1 ID: 1 T: 2018-09-19T09:08:39+02:00 - L1: 232.32V 0.46A NaNW NaNcos | L2: 232.45V 0.59A NaNW NaNcos | L3: 232.69V 1.62A NaNW NaNcos%!(EXTRA float64=NaN)
RTUClientHandler: 2018/09/19 09:08:39 modbus: received 01 04 04 42 46 74 58 28 d3
RTUClientHandler: 2018/09/19 09:08:39 modbus: sending "\x01\x04\x00\x0e\x00\x02\x10\b"
2018/09/19 09:08:39 SubscriptionManager: queued event: &{1537340919442 meterupdate DevID: 1, FunCode: 4, Opcode c: Value: 49.614}.
2018/09/19 09:08:39 UniqueId: Instrument1 ID: 1 T: 2018-09-19T09:08:39+02:00 - L1: 232.32V 0.46A 49.61W NaNcos | L2: 232.45V 0.59A NaNW NaNcos | L3: 232.69V 1.62A NaNW NaNcos%!(EXTRA float64=NaN)
RTUClientHandler: 2018/09/19 09:08:39 modbus: received 01 04 04 42 8f f6 5b d8 4c
2018/09/19 09:08:39 UniqueId: Instrument1 ID: 1 T: 2018-09-19T09:08:39+02:00 - L1: 232.32V 0.46A 49.61W NaNcos | L2: 232.45V 0.59A 71.98W NaNcos | L3: 232.69V 1.62A NaNW NaNcos%!(EXTRA float64=NaN)
RTUClientHandler: 2018/09/19 09:08:39 modbus: sending "\x01\x04\x00\x10\x00\x02p\x0e"
2018/09/19 09:08:39 SubscriptionManager: queued event: &{1537340919474 meterupdate DevID: 1, FunCode: 4, Opcode e: Value: 71.981}.
RTUClientHandler: 2018/09/19 09:08:39 modbus: received 01 04 04 43 bc 20 5d f7 dd
RTUClientHandler: 2018/09/19 09:08:39 modbus: sending "\x01\x04\x00\x1e\x00\x02\x11\xcd"
2018/09/19 09:08:39 SubscriptionManager: queued event: &{1537340919506 meterupdate DevID: 1, FunCode: 4, Opcode 10: Value: 376.253}.
2018/09/19 09:08:39 UniqueId: Instrument1 ID: 1 T: 2018-09-19T09:08:39+02:00 - L1: 232.32V 0.46A 49.61W NaNcos | L2: 232.45V 0.59A 71.98W NaNcos | L3: 232.69V 1.62A 376.25W NaNcos%!(EXTRA float64=NaN)
RTUClientHandler: 2018/09/19 09:08:39 modbus: received 01 04 04 3e ef 40 5c f6 60
RTUClientHandler: 2018/09/19 09:08:39 modbus: sending "\x01\x04\x00 \x00\x02p\x01"
2018/09/19 09:08:39 SubscriptionManager: queued event: &{1537340919541 meterupdate DevID: 1, FunCode: 4, Opcode 1e: Value: 0.467}.
2018/09/19 09:08:39 UniqueId: Instrument1 ID: 1 T: 2018-09-19T09:08:39+02:00 - L1: 232.32V 0.46A 49.61W 0.47cos | L2: 232.45V 0.59A 71.98W NaNcos | L3: 232.69V 1.62A 376.25W NaNcos%!(EXTRA float64=NaN)
[... cut ...]
RTUClientHandler: 2018/09/19 09:08:39 modbus: sending "\x01\x04\x00J\x00\x02P\x1d"
2018/09/19 09:08:39 SubscriptionManager: queued event: &{1537340919845 meterupdate DevID: 1, FunCode: 4, Opcode 164: Value: 0.000}.
2018/09/19 09:08:39 UniqueId: Instrument1 ID: 1 T: 2018-09-19T09:08:39+02:00 - L1: 232.32V 0.46A 49.61W 0.47cos | L2: 232.45V 0.59A 71.98W 0.53cos | L3: 232.69V 1.62A 376.25W 1.00cos%!(EXTRA float64=NaN)
RTUClientHandler: 2018/09/19 09:08:39 modbus: received 01 04 04 00 00 00 00 fb 8`

andig, 2018-09-19:
@timtimsson could you please open a new issue? This is getting out of hand ;) Also- as the *webpage* looks wrong- could you please check the browser console?

timtimsson, 2018-09-19:
finaly i clone git again - now all works fine - thanks!

PR #61 (2018-09-18): Publish connect message
Fix #59