gonium/gosdm630 #62: Add more DVG4013 features

andig, 2018-09-19:
The documentation seems to indicate that this meter supports more registers than currently implemented. It would be great to add those.

I would appreciate if anybody with a corresponding meter would get in contact.

ionfan, 2019-02-05:
i have this meter and i am very interested in improvements regarding more support for the DVG4013.
Especially the current wattage would be intersting to view.
What information do you need? I attached a screenshot from the vendor-software so that you can see which values can be read.
[image]

andig, 2019-02-05:
Ouch, seems they are defined in https://github.com/gonium/gosdm630/blob/master/meters/impl/dzg.go#L24 but not used. Will add asap.

frankrichter, 2019-02-05:
https://github.com/gonium/gosdm630/blob/master/meters/impl/dzg.go#L25 isn't
correct: 0x000 is Import (active) power and 0x0002 is Export (active) power.
All DZG meter reading are unsigned integers (uint32). To get a signed power
reading you have to calculate Import - Export.

>

andig, 2019-02-06:
> 0x000 is Import (active) power and 0x0002 is Export (active) power.

Right. This was never tested before, implementation was done blindly.

@ionfan would it be possible to get remote access to your meter, e.g. using ssh or vpn, for testing? Please pm me at cpuidle@gmx.de.