gonium/gosdm630 #111: Problem building from source

art0, 2019-01-10:
Thanks for gosdm360, trying to build from source.

When building from source ( go version go1.10.4 linux/amd64), it breaks with the following error

github.com/gonium/gosdm630/meters
../meters/measurements.go:126:10: m.String undefined (type *Measurement has no field or method String)
../meters/meter.go:91:55: iec.String undefined (type Measurement has no field or method String)

andig, 2019-01-10:
You need go 1.11 as of post 0.7

art0, 2019-01-10:
Thanks @andig Upgrading to 1.11 made the trick. Does 0.7 support the reading of Total Energy per phase?

andig, 2019-01-10:
Not implemented yet (SDM meter?) but simple to do :)

art0, 2019-01-10:
Well, it seems like total "Import" (Kwh) is there per phase in a SDM
 "Import": {
  "L1": 29.529998779296875,
  "L2": 47.463996887207031,
  "L3": 72.39898681640625
 },

andig, 2019-01-11:
Indeed, done for import/export but not for "net":

		ImportL1:  0x015a,
		ImportL2:  0x015c,
		ImportL3:  0x015e,
		Import:    0x0048,
		ExportL1:  0x0160,
		ExportL2:  0x0162,
		ExportL3:  0x0164,
		Export:    0x004a,

I guess this can be closed then?

art0, 2019-01-11:
Hi @andig, what do you mean by "net"?

I made a small patch to include the Maxium Total Power and the Maximum Current per phase

       MaxTotalPower: 0x0056,
                MaxCurrentL1: 0x0108,
                MaxCurrentL2: 0x010A,
                MaxCurrentL3: 0x010C,

       Frequency   *float64
        MaxTotalPower   *float64
        MaxCurrent ThreePhaseReadings
}

Not sure if this data needs to be supported in all meters to be added?

andig, 2019-01-11:
New readings require changes to the http cache that I still haven‘t come around to actually implement. Good if working for you!

closed 2019-01-13 by andig