gonium/gosdm630 #114: Connect few meters to one dongle

poratnir1, 2019-01-22:
Is it possible to read two meters using same service ?
or using same converter?

or need to add two services?

J-A-U, 2019-01-22:
As the name says Modbus is a communication-bus. On a bus may a bunch of clients be connected to one converter, each with a different address. You tell gosdm the devices on the bus with paramter -d. For example
`-d SDM:22,SDM:23`
two Eastron SDM-meters, one with adress 22, one with 23.

andig, 2019-01-22:
You can connect multiple meters but they must have different modbus ids.

andig, 2019-01-24:
I feel this has been answered. If not please reopen.

closed 2019-01-24 by andig

Poratnir, 2019-01-25:
Thanks!! working great

what speed is recommended?

andig, 2019-01-25:
The exact speed that your meter requires!

J-A-U, 2019-01-25:
Well, you have to take care alle meters on one bus are on the same baud- and framerate.
I suggest al least 9600bd. The SDM120 are delivered with 2400bd, thats pretty slow.