gonium/gosdm630 #128: Failed to connect: no such file or directory

dimagoltsman, 2019-02-28:
using the arm release.

cant be started, it just throws `Failed to connect: no such file or directory`  without any explanations

andig, 2019-02-28:
Hard to guess. Which release? How to you invoke it?

dimagoltsman, 2019-02-28:
latest, 0.70
```
pi@hass-pi:~/gosdm630/bin $ ./sdm_detect 
2019/02/28 14:38:12 Failed to connect: no such file or directory

pi@hass-pi:~/gosdm630/bin $ /home/pi/gosdm630/bin/sdm630 -u 0.0.0.0:63080 --broker tcp://127.0.0.1:1883 --user homeassistant --password dIMAGO86 --qos 0 --clean --rate 15
2019/02/28 14:38:31 Failed to connect: no such file or directory

```

frankrichter, 2019-02-28:
You don't use -a and -d options. Do you have a serial adapter connected to
/dev/ttyUSB0?

dimagoltsman, 2019-02-28:
yeah, i use this one https://www.ebay.com/itm/CH340-USB-to-RS485-485-Converter-Adapter-Module-For-Win7-Linux-XP-Vista/201258967189?epid=17006943744&hash=item2edbf82095:g:bPgAAOSw7XZXhzJe:rk:3:pf:1

dimagoltsman, 2019-02-28:
oh damn.... u r right... its just changed to USB1 have no idea why....

closed 2019-02-28 by dimagoltsman