gonium/gosdm630 #131: Wrong values with sdm530

RubberDucky007, 2019-03-16:
I got an Digitus DA-70157 RS-485 Adapter and a Raspi 1b with volkszähler-latest and gosdm630 binary (V.6.0 shown) on an SDM530 Device. It seems to work technically but most of the readings are zero or Totally wrong.
It seems to me sdm630 reaches only the values from L1 an L2 and copies L2 to L3 values.
The total import kwh is doubled, the total Watts shown are L1+L2+L2 because the values on the Meter display are different to what sdm630 shows. I don't know what i can do to fix this. Need Help.
[image]

andig, 2019-03-16:
You‘re apparently running 0.6.0 which is really old? Could you run current git version and check again?

RubberDucky007, 2019-03-16:
I believed i got the latest 0.7.0 here, but also believe it shows itself as 0.6.0. Could it be the error is with packaging for arm-platform? I will test again with an fresh downloaded 0.7.0.

andig, 2019-03-16:
Could you just compile current master branch?

RubberDucky007, 2019-03-16:
Hallo. Ich wechsele mal auf Deutsch weil es einfacher ist. Ja, ich hab den aktuellen master branch geklont aber auf dem raspi 1b hier fehlen wohl tools zum kompilieren. zip konnte ich noch nachinstallieren doch jetzt bekomme ich beim aufruf von ./build.sh 

`Building for ...
... Linux/32bit
can't load package: package _/home/pi/gosdm630/cmd/sdm: cannot find package "_/home/pi/gosdm630/cmd/sdm" in any of:
        /usr/lib/go-1.7/src/_/home/pi/gosdm630/cmd/sdm (from $GOROOT)
        ($GOPATH not set)
        zip warning: name not matched: /home/pi/gosdm630/bin/*

zip error: Nothing to do! (/home/pi/gosdm630/release/sdm-linux-386.zip)
... Linux/64bit
can't load package: package _/home/pi/gosdm630/cmd/sdm: cannot find package "_/home/pi/gosdm630/cmd/sdm" in any of:
        /usr/lib/go-1.7/src/_/home/pi/gosdm630/cmd/sdm (from $GOROOT)
        ($GOPATH not set)
        zip warning: name not matched: /home/pi/gosdm630/bin/*

zip error: Nothing to do! (/home/pi/gosdm630/release/sdm-linux-amd64.zip)
... Raspberry Pi/Linux
can't load package: package _/home/pi/gosdm630/cmd/sdm: cannot find package "_/home/pi/gosdm630/cmd/sdm" in any of:
        /usr/lib/go-1.7/src/_/home/pi/gosdm630/cmd/sdm (from $GOROOT)
        ($GOPATH not set)
        zip warning: name not matched: /home/pi/gosdm630/bin/*

zip error: Nothing to do! (/home/pi/gosdm630/release/sdm-linux-arm.zip)
... MacOS/64bit
can't load package: package _/home/pi/gosdm630/cmd/sdm: cannot find package "_/home/pi/gosdm630/cmd/sdm" in any of:
        /usr/lib/go-1.7/src/_/home/pi/gosdm630/cmd/sdm (from $GOROOT)
        ($GOPATH not set)
        zip warning: name not matched: /home/pi/gosdm630/bin/*

zip error: Nothing to do! (/home/pi/gosdm630/release/sdm-darwin-amd64.zip)
... Windows/32bit
can't load package: package _/home/pi/gosdm630/cmd/sdm: cannot find package "_/home/pi/gosdm630/cmd/sdm" in any of:
        /usr/lib/go-1.7/src/_/home/pi/gosdm630/cmd/sdm (from $GOROOT)
        ($GOPATH not set)
        zip warning: name not matched: /home/pi/gosdm630/bin/*

zip error: Nothing to do! (/home/pi/gosdm630/release/sdm-windows-386.zip)
... Windows/64bit
can't load package: package _/home/pi/gosdm630/cmd/sdm: cannot find package "_/home/pi/gosdm630/cmd/sdm" in any of:
        /usr/lib/go-1.7/src/_/home/pi/gosdm630/cmd/sdm (from $GOROOT)
        ($GOPATH not set)
        zip warning: name not matched: /home/pi/gosdm630/bin/*

zip error: Nothing to do! (/home/pi/gosdm630/release/sdm-windows-amd64.zip)
`

Wobei ich leider noch nicht mal eine Ahnung habe was ich da starte. Dem Output nach will das alle ziele bauen, obwohl ich nur linux-arm6 brauche (Raspi 1b).

Ein fertiges binary wäre wirklich einfacher für mich.

andig, 2019-03-16:
Du brauchst mitdestens go 1.11- leider von Hand zu isntallieren :/

RubberDucky007, 2019-03-16:
Ist installiert.

`pi@rpi1b:~ $ apt show golang
Package: golang
Version: 2:1.7~5
Priority: optional
Section: devel
Source: golang-defaults
Maintainer: Go Compiler Team <pkg-golang-devel@lists.alioth.debian.org>
Installed-Size: 10,2 kB
Depends: golang-1.7, golang-doc (>= 2:1.7~5), golang-go (>= 2:1.7~5), golang-src (>= 2:1.7~5)
Homepage: https://golang.org
Download-Size: 3.538 B
APT-Manual-Installed: yes
`
keine broken dependencies, golang[-go|src] sind auch installiert. 
Ist der aufruf von ./build.sh denn richtig oder wie starte ich das kompilieren für arm im verzeichnis?

RubberDucky007, 2019-03-16:
Sorry, hab erst jetzt README.md gefunden wo es heißt ich brauche auch noch go-dep und embed. Doch leider ist das hier ein volkszähler-image das man nicht einfach so updaten soll (heißt es) und go-dep braucht golang 1.9 und ich hab hier 1.7. Im raspian repo sind beide nicht und kompilieren scheitert. Damit bin ich am ende meiner Weisheit und kann dies vermutlich zu den akten legen.

closed 2019-03-16 by RubberDucky007

andig, 2019-03-16:
Wie gesagt:

> Du brauchst mitdestens go 1.11- leider von Hand zu isntallieren :/

Einfach auf der go Seite schauen, ist nicht schwierig ;)

frankrichter, 2019-03-16:
Notizen von meiner Go-Installation vor ein paar Wochen:

cd ~
sudo apt remove golang
sudo apt autoremove
wget https://dl.google.com/go/go1.11.4.linux-armv6l.tar.gz
sudo tar -C /usr/local -xvf go1.11.4.linux-armv6l.tar.gz 
mkdir go
echo 'export GOPATH=$HOME/go' >> ~/.profile
echo 'PATH="$PATH:/usr/local/go/bin:$GOPATH/bin"' >> ~/.profile
source .profile