gonium/gosdm630 #133: GPIO to switch between RX/TX

3s1d, 2019-04-17:
As I am a little paranoid I wanted to have a galvanicaly isolated RS485 bus. (The ends in the SDM630 are very close to the hot wires...) I bought the only solution for the Raspberry Pi I could find: https://www.hwhardsoft.de/deutsch/projekte/rs485-shield/
NB: The FTDI seems to break the USB hub on the Raspberry Pi after some time.

The problem is the the auto send/receive control on this board is BS (as it's based on an NE555). I need to have GPIO 18 high for TX and low for RX, respectively.

Could you please add GPIO support for the Raspberry Pi to do this? Unfortunately, I have zero Go experience...

Thanks

andig, 2019-04-17:
Not really as this is all part of the underlying library, sorry.

closed 2019-04-17 by andig

3s1d, 2019-04-19:
Just for completion: I managed to get it working w/ auto TX/RX by changing the default hardware TX timeout from 40uS to approx 1ms (by replacing the 3k9 resistor w/ a 100k one).
Between TX and RX of the SDM630 are at least 4ms, so there is plenty of time for switching...

Cheers,
Juergen