gonium/gosdm630 #140: Building on RPi fails

suelberg, 2019-05-27:
Hej! I am trying to build the latest code on a RPi.

The following error I cannot resolve:

```
Version: 05a7865 2019-05-27_15:01:08
go build -v -ldflags '-X "sdm630.Version=05a7865" -X "sdm630.Commit=05a7865"' ./...
github.com/gonium/gosdm630/meters
# github.com/gonium/gosdm630/meters
meters/measurements.go:164:10: m.String undefined (type *Measurement has no field or method String)
meters/meter.go:91:55: iec.String undefined (type Measurement has no field or method String)
Makefile:20: die Regel für Ziel „binaries“ scheiterte
make: *** [binaries] Fehler 2
```

I tried two version of Go (1.11 & 1.12) and use the "make binaries" as golangci-lint seems to be unavailable on Arm. 

Any suggestions?
Thanks a lot!

andig, 2019-05-27:
Please try

    make

Or

    make build

Both should work by creating the requires files in the process.

suelberg, 2019-05-28:
Thank you very much, that worked!

closed 2019-05-28 by suelberg